frangoteam/FUXA #1320: Accepts mq messages to control the node values of another device

ZXiaoPang, 2024-08-14:
FUXA version：1.2.0-1815

[image]
This image shows the commands received from the topic I subscribed to.

[image]
How can I modify the value of a node on another device in the image at the same time I receive this command?

closed 2024-08-15 by ZXiaoPang

haslish, 2024-08-15:
@ZXiaoPang Did you manage to get your script to show up in the dropdown list?

ZXiaoPang, 2024-08-19:
> @ZXiaoPang Did you manage to get your script to show up in the dropdown list?

I can't get the script to appear in the list.