frangoteam/FUXA #1321: [BUG]SiemensS/ Plugin

somiom, 2024-08-14:
Dear frangoteam,

After i updated Fuxa from version 1.1.19 to version 2.0.0., connection to S7 stop to work. In logs I saw that Siemen S7 plugin is mising.

[image]

**Environment**
 - FUXA version: 1.2.0
 - Node.js version: v22.6.0
 - npm version:
 - Platform/OS: Ubuntu / docker container
 - Browser: Chrome

unocelli, 2024-08-14:
Hi, in docker try frangoteam/fuxa:snap7-1.2.0

AlexCastro-Prog, 2024-08-14:
Hi Frango team,

I installed on docker and I didn't see the Rockwell Automation node or connection to be added.

I have 7 rockwell PLC's that I need add on Fuxa but I didn't where the connection to connect rockwell plc.

MatthewReed303, 2024-08-15:
@AlexCastro-Prog have you gone into plugins and then press the + next to EthernetIP?

AlexCastro-Prog, 2024-08-15:
Yes. I got it. Thank you!

Other question is: Is it possible to update for a new version of fuxa with the current project?

MatthewReed303, 2024-08-15:
@AlexCastro-Prog you cab use save project as and this will download a json of your current project, then open new version of Fuxa and open project and wait, if it's a large project it takes a little time.

unocelli, 2024-08-17:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-08-17 by unocelli

jmcastillejo, 2025-04-26:
> Dear frangoteam,
>
> After i updated Fuxa from version 1.1.19 to version 2.0.0., connection to S7 stop to work. In logs I saw that Siemen S7 plugin is mising.
>
> [image]
>
> **Environment**
>
> * FUXA version: [...]

snap7 driver not installed in latest docker version. Install with: "docker exec -it <CONTAINER ID> npm i node-snap7"