frangoteam/FUXA #1324: [FEATURE]Is there way to bind object property to a dynamic tag?

jay-jjwu, 2024-08-20:
**Describe the feature**
Is there way to bind object property to a dynamic tag?

**Describe the solution you'd like**
Describe your proposed solution here.

For example, i have an INPUT object "input_1"binding to a specific tag "tag_1", 
and another object "input_2" binding to "tag_2"....
then "input3" to "tag_3"......etc

In this way, i would like to have the input_X binding to tag_X, based on the object name, instead of a static tag name. Is there a way to have a dynamic tag name typing into the property tab fields...for example something like "tag{ getTagName() }". Or the tag filling can be filled by javascripts. In this way to have the tag binding dynamicly.

**Additional context**
If applicable, add screenshots to help explain your idea.
[image]

henjoe, 2024-09-03:
Use Placehoder instead. 

Instead of binding it to a specific Tag, just simple put @Input1, @input2, @input3 and etc. 

And when you call the Page, simply bind specific tag for every place holder.

unocelli, 2024-10-27:
Let's close here, since there is nothing feasible

closed 2024-10-27 by unocelli