frangoteam/FUXA #1328: Feature Request: Support for Differential Data Calculation Based on Time Units in Bar Chart Module

Fragile-boy, 2024-08-21:
Hello, first of all, thank you for open-sourcing this excellent project.

### Background:

Our data is accumulated in registers, and the values continuously increase with production.
（ Honestly, I'm not very clear on the functions of the three buttons shown below; if you could also explain those, I would be very grateful.）
[image]

### Requirement:

We want to see the production volume, number of faults, fault handling time, etc., on an hourly and daily basis. Therefore, what we need is the differential value of the data over time units.
[image]
The effect shown above is achieved by saving the total daily production in several registers. Now, we want to view the hourly production totals, but using 24 registers for this purpose feels a bit inelegant.

### Current Situation:

In the bar chart module, after selecting the "data and time" option for the X-axis, the only available data operations are "hour sum" and "sum," which is clearly not what I need.
[image]

Is there a feature that I might have overlooked? Please let me know, so I can implement it to meet my needs.

Fragile-boy, 2024-08-21:
It seems that #1283 solves my problem perfectly. I will update the version and try again.

Fragile-boy, 2024-08-23:
Today I noticed that when using SQLite as the database, after terminating the project with ctrl+C, the $getHistoricalTags function can no longer retrieve historical data. However, when using FluxDB, it continues to work.

Are you aware of this issue?

unocelli, 2024-08-23:
Hi, do not mix issues, influxdb is a motor that work autonomously, Sqlite is a files based database, fuxa will be the interface to the files, if you stop it you have to read yourself the files located in _db

Fragile-boy, 2024-08-24:
Hi, thank you for your response. Here is my understanding of your answer:

When using SQLite as the database, every time Fuxa is started, it creates a new DB file where the data is stored, and it can only interact with the data in this DB file.

When using InfluxDB as the database, regardless of the situation, there is only one database, and all operations are conducted solely within this configured database.

Is my understanding correct?

henjoe, 2024-09-03:
> It seems that #1283 solves my problem perfectly. I will update the version and try again.

Did you manage to do this using the script? I am looking for similar things. How do you populate the integral values from the Scripts results in the Bar chart?

unocelli, 2024-10-27:
Let's close here, since there is nothing feasible

closed 2024-10-27 by unocelli