frangoteam/FUXA #1331: Read scale script

haslish, 2024-08-23:
How does the read scale script work? I cant choose i script the drop down menu. 
[image]

Is it possible to set the timestamp in script? I have a water meter that send reports 4 times a day and the data contains numbers for each hour. So i have 8 values that i would like to write with different timestamps at the same time.

haslish, 2024-08-24:
I found out how to write a script that I can choose in the read scale script but i dont understand how to return a value to the tag
The tag value is  "tC": 44.6,"tF": 112.3 and i want to return 44.6 to the tag. I only managed to write a scheduled script that writes the value to a different tag.

[image]

[image]

henjoe, 2024-09-03:
> I found out how to write a script that I can choose in the read scale script but i dont understand how to return a value to the tag The tag value is "tC": 44.6,"tF": 112.3 and i want to return 44.6 to the tag. I only managed to write a scheduled script that writes the value to a different tag.
> 
> ![script 2](https://private-user-images.githubusercontent.com/104933222/361104379-c570bd34-4f4a-4ea3-8e0e-762357f1a6c3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUzNjgyMjIsIm5iZiI6MTcyNTM2NzkyMiwicGF0aCI6Ii8xMDQ5MzMyMjIvMzYxMTA0Mzc5LWM1NzBiZDM0LTRmNGEtNGVhMy04ZTBlLTc2MjM1N2YxYTZjMy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwM1QxMjUyMDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wMDVkYjMzY2M4YTVkNTljNDk3YWJhZWFhNDM3NzY4NDkwMzM0MmYzY2FjNGNjMzI1NjczZWQ5NjMwZDI0OGI1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.leKM1x192oCG9D7X5oszAmALMShkjNUaSvLCcvjDP7Q)
> 
> ![script](https://private-user-images.githubusercontent.com/104933222/361104412-5bfe738d-6e23-4761-97f0-30425ef5bd6c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUzNjgyMjIsIm5iZiI6MTcyNTM2NzkyMiwicGF0aCI6Ii8xMDQ5MzMyMjIvMzYxMTA0NDEyLTViZmU3MzhkLTZlMjMtNDc2MS05N2YwLTMwNDI1ZWY1YmQ2Yy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwM1QxMjUyMDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yYjk3OWI1NWVmNmI5NGE1YmYxZGU4MWE4NTdkM2ZhYzQ1YjI5NDkwM2UwNWUwZTM1ZWJhYTBmM2E0N2FiODhiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.50eWlXL_noKY1ymKmhwsgiUoPySsj312rHk8742JjyM)

try this
```
const outValue = value['tC']
return outValue
```

and always remember to add the "value" parameter on your scripts.

haslish, 2024-09-03:
Didnt get it to work with @henjoe suggestion.

Can it have something to do with the json format? I tried to write values, strings and json strings to it with the read scale script but the value doesnt change from [object Object]

[image]

unocelli, 2024-09-11:
for Read Scale Script and Write Scale Script the script must always return a value. the @henjoe proposal is correct

haslish, 2024-09-13:
What am I missing?

[image]

[image]

jay-jjwu, 2024-09-14:
@haslish ， hi, how did you add the script into selection list? i stil have no way to select after adding acripts. 
[image]

haslish, 2024-09-14:
[image]
When you add a value as script parameter the script shows up in the selection list.

jay-jjwu, 2024-09-14:
@haslish 
Many thanks, it did work!
i never expect the fuxa system designed as this special way....:-)

unocelli, 2024-10-27:
Let's close here, since there is nothing feasible

closed 2024-10-27 by unocelli