frangoteam/FUXA #1343: [BUG]_readMemery Error When Enabling Multiple Devices with 150 Tags Each

Fragile-boy, 2024-09-02:
**Describe the bug**
I have configured two devices, each with approximately 150 tags. 
[image]
[image]

When I enable either device individually, Fuxa works normally without any issues. However, when I enable both devices simultaneously, the log persistently reports errors.

[image]

I have already tried the solutions mentioned in issues #154 and #191, but unfortunately, they didn't resolve the problem. The errors still occur when both devices are enabled at the same time.

Additionally, I attempted to enable the fragmented flag, but this caused another issue: the application was unable to read all tag values properly.
[image]

Could you please provide further assistance or suggest any additional steps I can take to resolve this issue?

Thank you!

**Environment**
 - FUXA version: 1.2.0
 - Node.js version: 
 - npm version:
 - Platform/OS: docker
 - Browser:chrome 128.0.6613.114

unocelli, 2024-09-03:
Hi, have you try the version in master branch with PR #1312 ?

Fragile-boy, 2024-09-04:
> Hi, have you try the version in master branch with PR #1312 ?

Hi, thanks! I followed your instructions and after applying the update, it perfectly solved the problem.

Fragile-boy, 2024-09-09:
> > Hi, have you try the version in master branch with PR #1312 ?
> 
> Hi, thanks! I followed your instructions and after applying the update, it perfectly solved the problem.

I'm sorry, this issue still exists. Below are my relevant settings.

The process is like this: I first closed all connections. Then, I opened the first connection on the left. After I was able to stably see the data, I turned on the Socket reuse switch, and then I started the second one.

After that, the second connection and the first one began alternating in updating data, but only part of the data could be updated, and the console started continuously reporting various errors, such as 'read_memory Error,' or 'length too long, expect 87 got 107,' and similar ones.
[image]
[image]
[image]
[image]

unocelli, 2024-09-19:
@rikugun Hi, do you have an idea of the issue?

Fragile-boy, 2024-09-23:
> @rikugun Hi, do you have an idea of the issue?

Sorry for the late reply.

The background of my issue is that I have a serial server connected to multiple PLCs via RS485. The PLCs only support Modbus RTU communication, so the serial server performs Modbus TCP to RTU conversion.

Although socket reuse shares a single connection, RTU is half-duplex communication, and the serial server does not implement a message queue. As a result, when two Modbus TCP requests arrive at the serial server almost simultaneously, one of them is discarded. This leads to issues with some updates not being processed correctly. (Later, I used Python to create my own message queue, simulating requests sent from FUXA, and all requests received correct responses.)

My suggestion:
Add a new option, whether to enable the message queue function. For the same socket, the next request would only be sent after receiving the response for the previous request. (Our situation is somewhat special, so I think this might not be very meaningful.)

rikugun, 2024-09-23:
I have a similar problem when use rtu to tcp , I'm trying to solve.

rikugun, 2024-09-27:
hi @Fragile-boy ,  I make a read/write mutex when use  socket reuse , I'd tested with my devices, can you help me to  test your devices on my branch 

https://github.com/rikugun/FUXA/tree/feat/modbus-tcp-mutex

[image]

Fragile-boy, 2024-10-08:
Thank you for your hard work. I apologize for the delay, but I have transitioned to using a cloud platform with secondary development to meet my needs.

In the case of FUXA, during communication with PLCs, it can only act as a client while the serial port server acts as a server. This means the IP address of the serial port server must remain fixed. However, in the production line, after the machines are sold, the IP addresses often change. This is the fundamental reason why I chose to develop a custom solution.

Additionally, I realized that what I need is a data reporting display rather than real-time interaction with a SCADA-like application (even though FUXA could fulfill this, I felt it didn’t quite fit my specific needs, like implementing Gantt charts).

Overall, FUXA is a great project, and its scripting module gives it unlimited potential. I'm sorry for the delayed response, and I will test it as soon as possible. Once I have the results, I will reply with my findings.@rikugun

Fragile-boy, 2024-10-11:
Hi，由于另外一个服务高频使用串口服务器的原因，我再使用fuxa连接串口服务器，虽然能够连接成功，但是并没有获取到数据。所以我没有能够验证他是否能够串行的获取数据。对此我很抱歉

Fragile-boy, 2024-10-11:
Hi, due to another service frequently using the serial port server, although I was able to connect to the serial port server using FUXA, I couldn't retrieve any data. As a result, I wasn't able to verify whether it can successfully obtain serial data. I apologize for this.

unocelli, 2024-10-27:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-10-27 by unocelli

henjoe, 2024-11-01:
Hi I dont know what is the issue here but I tried the same setup I have 2 Modbus serial slaves connected to a modbusTCP gateway where Fuxa connected to it. Something like this below:

[image]

And it works fine for me, I can connect to both Slave IDs 1 & 2 for modbusTCP gateway (192.168.1.1), I just simply create 2 device connections and add their slaveIDs accordingly, (very similar on what you did). And I view both the data. 
The thing is I didn't use the reuse socket cause I am on the old version (V1.2.0) and I can confirm that it works fine to me.

PR #1412 (2024-10-01): Feat/modbus tcp mutex
fix #1343

For modbus-tcp,  if open socket reuse ,  add option to read/write serial.

[image]