frangoteam/FUXA #1345: [BUG]SiemensS7协议打包exe或者app后连接不上

yu-1949, 2024-09-02:
使用electron打包exe和打包app后。SiemensS7协议连接不上。也没有显示连接状态。在日志中看到【warn】 try to create test but plugin is missing! 请问这个是哪里的问题

rikugun, 2024-09-08:
S7 not installed by default ,try install first

 `npm install node-snap7;`

yu-1949, 2024-09-09:
> S7 默认未安装，请先尝试安装
> 
> `npm install node-snap7;`

Hi, I tried npm install node-snap7 locally and it works. Problem after packaging as exe. After the function is enabled, the failure and success status are not displayed

imic, 2024-11-14:
This issue is related to packaging an Electron application. The electron-rebuild should recompile the snap7 native module to be compatible with the Electron version. However, I noticed that the packaged files still contain the native modules used by Node.js, rather than the Electron-specific modules. I tried manually rebuilding, and the snap7 module worked, allowing me to connect to the S7 device. But as soon as I tried to read data, the entire Electron application crashed.

s5314203, 2025-01-24:
Has this not been resolved yet?

fengchenyu, 2025-04-27:
This problem does exist, and I have also encountered it

jingshui127, 2025-05-07:
最新的1.2.5版本，这个问题也依然存在！

jingshui127, 2025-05-12:
打包成electron后S7都无法正常使用。
我测试是最新的1.2.5版本。
并且在WIN10/11和Windows Server 2022都进行了测试。

rikugun, 2025-05-12:
提供下错误信息，或者使用未打包的版本测试下情况。我这没有使用S7的环境。

jingshui127, 2025-06-18:
未打包，直接在Server运行，node main.js是正常的。electron打包之后，Snap7也打包进去了。但还是报错【warn】 try to create test but plugin is missing!