frangoteam/FUXA #1351: [BUG]

xingjl86, 2024-09-03:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'DAQ storage'
2. Choose Tdengine for storage and the tag registored.
[image]
[image]

4. See error
The following alarm was add to the log file
2024-09-03T06:22:04.484Z [error] daq addValue error FetchError: request to http://localhost:6041/rest/sql/ failed, reason: connect ECONNREFUSED 127.0.0.1:6041
**Expected behavior**
Is it about the JDBC driver of tdengine? What should I do to make this OK？Reinstall the driver?

**Screenshots**
[image]
[image]

**Environment**
 - FUXA version: latest
 - Node.js version: 16.20.2
 - npm version: 8.19.4
 - Platform/OS: Ubuntu
 - Browser: firefox

xingjl86, 2024-09-03:
The 6041port have been tested and confirmed OK via Grafana and curl command.
taos$ curl -L -H "Authorization: Basic cm9vdDp0YW9zZGF0YQ==" -d "select rrr, ppp from my_table" localhost:6041/rest/sql/test
{"code":0,"column_meta":[["rrr","TIMESTAMP",8],["ppp","INT",4]],"data":[],"rows":0}
[image]

eduardoexact, 2024-09-11:
I had an issue with tdengine also. I had a mqtt conection and I'm saving some values in DAQ Storage. After looking into the tdengine databases I realized that fuxa wasn´t crating the tables sucessfully. I managed to work with tdengine after renaming the mqtt conection id from: `d_766bb05b-71d74601` to `d_766bb05b_71d74601` in the json project. I think we cant use the `-` character for the tdengine table.

xingjl86, 2024-10-14:
How can we help if it were a bug?

xingjl86, 2024-11-11:
I have solved this
 problem by change the "localhost" into the IP of the device. For example,	the IP of my PC is 10.100.1.10, i just set "http://10.100.1.10:6041/rest/sql" instead of "http://localhost:6041/rest/sql" or "http://127.0.0.1:6041" even though the FUXA and TDEingine are installed on the same local PC, then it works.

closed 2024-11-11 by xingjl86