frangoteam/FUXA #1354: [FEATURE] Request for addition of mouse event 'Hover'

Kiruthick-Roxx, 2024-09-04:
**Description**
We would like to request a new feature that adds hover functionality to the gauge components in the Shape settings (Events). This enhancement would provide additional interactivity by allowing users to view extra details, change styles, or trigger actions when hovering over gauge elements.

**Describe the solution you'd like**
[image]
In the dropdown box, one more event can be added like " Hover" which will check for mouse events using the mouseover or mouseout fucntionality or something better than that so that when we move the mouse over the object, it creates a transition or expose addtional information about the object. It can also be used to trigger a script.

**Additional context**
If there are any existing plugins, third-party tools, or similar implementations that could serve as a reference for this feature, please feel free to share them.

mauroalexandre, 2024-09-04:
Hi @Kiruthick-Roxx 

Sugestion are implemented here https://github.com/frangoteam/FUXA/pull/1355

Thanks

Kiruthick-Roxx, 2024-09-04:
Hi mauroalexandre,
    Thanks for the quick response. I actually forgot to check your forks. Thank you once again.

kanaCV, 2024-09-04:
Hi @mauroalexandre,

To add to the request, is it possible to create transition of the same image which has been assigned with the mouseover and mouseout event from the scripts? 

For example:
If i move the mouse over an image, it should expose few more options by a transition or i when i move over the image, it should make some animations of the same image.

If this is possible can you kindly explain how it can be done along with an example script since, I'm entirely new to fuxa as well as scripting in JS.

MatthewReed303, 2024-09-04:
@kanaCV dynamic SVGs are in the works where you can use script tags and css within the SVG file and you can do what ever effects you want using scripting and css. Currently @unocelli is working on binding of tags for the SVG attributes. To get an idea how this will work google dynamic svgs and animating svgs etc

mauroalexandre, 2024-09-06:
@MatthewReed303 @kanaCV 

I created an example of how to manipulate SVGs using scripts. I'm not sure if this meets your exact needs, but I hope it helps.
[Fuxa_Manipulate_SVG-Script_Example.json](https://github.com/user-attachments/files/16909976/Fuxa_Manipulate_SVG-Script_Example.json)

@MatthewReed303 @unocelli In the future, it would be useful to consider the possibility of retaining the ID properties of the paths from the original SVG when it is imported. This would greatly facilitate this type of manipulation.

Kiruthick-Roxx, 2024-09-12:
Hi @mauroalexandre 
Thanks for the example. I'll try this and let you know.

Kiruthick-Roxx, 2024-09-14:
Hi @mauroalexandre
Regarding the example that you gave, you have changed the colour of the text elements that is available in the entire page. My application would be to animate a specific svg with an ID that uses path element instead of shapes like circles and rectangle or texts and only animate that particular svg element with a script particular to that element only.

For exmple, the one i attached below.
[image]

I want to know how to to script the JS to do this for individual SVG element which is of PATH type. If I want to use some other types of Svg to make it possible, Please let Me know. 

[image]
Also I have other ideas to implement this kind of animates that expose other option using mouse events. Still I'm very new to fuxa scripting and how to link to elements individually. Sorry if I'm too dump to ask this kind of questions. But this would help many people like me to make the dashboards more interactive and feature rich.

Thanks,.

mauroalexandre, 2024-09-14:
@Kiruthick-Roxx 

Unfortunately, it seems that the path ID is not retained when imported into Fuxa. I believe this might be a deeper issue related to the editor used in the project. I will try to investigate further and see if I can contribute in this regard as soon as possible.

closed 2024-10-27 by unocelli