frangoteam/FUXA #1361: [BUG] Siemens String to OPCUA

LAB-KN, 2024-09-05:
**Describe the bug**
If the datatype "String" is selected in Siemens PLC OPC UA server, it will not show datatype in Fuxa OPC UA Client
If datatype is changed to WString in PLC, Fuxa can see it as a String.

**To Reproduce**
1. Setup OPC UA in Siemens PLC (via TIA portal)
2. Set a Tag as datatype "String"
3. Add Device  "OPCUA" in Fuxa
4. Find PLC OPC server Interface, and look for String Tag
5. String Tag will be missing Datatype

**Expected behavior**
Expected to see datatype for String

**Screenshots**
[image]
[image]

**Environment**
 - FUXA version: v1.2.0
 - Platform/OS: Windows 11
 - Browser: Chrome
 - Siemens Tia: v17 & v19

unocelli, 2024-09-11:
Hi, I can't reproduce it.
[image]

LAB-KN, 2024-09-20:
I have found that this is a siemens plc issue, i will close this as it is not a Fuxa bug.

[image]

closed 2024-09-20 by LAB-KN