frangoteam/FUXA #1363: [BUG] View Background Color not working

LordNetro, 2024-09-09:
**Describe the bug**
The background color and behaviour of the view is lost whenever you acces the view by an event.
_If you acces the view via the side menu the behaviour is the expected (it works). The bug only happens when you acces the view via a button in a different view_

**To Reproduce**
Steps to reproduce the behavior:
1. Add a button to a view
2. Add an **mouse down** / **open page** event to the button
3. Assign a view that has a different background color than the default to the event
4. Try it and see that the background color is white and not the one that you assigned before

**Expected behavior**
See the view with the expected background color and behaviour.

**Screenshots**
View accessed via button in different view
[image]

View accesed via side menu
[image]

**Environment**
 - FUXA version: 1.2.0-1751
 - Node.js version: v18.20.3
 - npm version: 10.7.0
 - Platform/OS: Windows 10 Pro
 - Browser: Microsoft Edge

henjoe, 2024-09-12:
I checked the logic for this, the sideNav has different logic from the button (navigating to other page). 

If you can edit a code, you can simply change the **fuxa-view.component.ts** under client\src\app\fuxa-view

Search the code below: (loadPage)
```
 loadPage(param: any, viewref: string, options: any) {

        let view: View = this.getView(viewref);
        if (view) {
            if (options?.variablesMapping) {
                this.loadVariableMapping(options.variablesMapping);
            }
            this.loadHmi(view);
            if (param.scaleMode) {
                Utils.resizeViewRev(this.dataContainer.nativeElement, this.dataContainer.nativeElement.parentElement?.parentElement, param.scaleMode);
            }
        }
    }
```

and change the _this.loadHmi(view)_ to _this.loadHmi(view, true)_
```
loadPage(param: any, viewref: string, options: any) {

        let view: View = this.getView(viewref);
        if (view) {
            if (options?.variablesMapping) {
                this.loadVariableMapping(options.variablesMapping);
            }
            this.loadHmi(view, true);
            if (param.scaleMode) {
                Utils.resizeViewRev(this.dataContainer.nativeElement, this.dataContainer.nativeElement.parentElement?.parentElement, param.scaleMode);
            }
        }
    }
```

henjoe, 2024-09-12:
By the way, how you manage all the controls (sliders, gauges and etc.) to have a little bit of transparent background? 

This is awesome.

henjoe, 2024-09-12:
> By the way, how you manage all the controls (sliders, gauges and etc.) to have a little bit of transparent background?
> 
> This is awesome.

> By the way, how you manage all the controls (sliders, gauges and etc.) to have a little bit of transparent background?
> 
> This is awesome.

Nevermind, I knew it. What a dumb question 😄

LordNetro, 2024-09-16:
> I checked the logic for this, the sideNav has different logic from the button (navigating to other page).
>
> If you can edit a code, you can simply change the **fuxa-view.component.ts** under [...]

Thank you so much!

unocelli, 2024-09-19:
@henjoe Thanks a lot!

unocelli, 2024-10-27:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-10-27 by unocelli