frangoteam/FUXA #1365: [BUG] Writing with scale script

timmpo, 2024-09-09:
**Describe the bug**
The function to have scale script is a verry nice thing, a must have when working with time schedules.

Read with scale script works fine but when i try to write with scale script i got :
ERR setValue script error, returned buffer invalid must be mod 2

mode 2 is what i can understand execute i server, that setting is set

**To Reproduce**
Steps to reproduce the behavior:
1. fire up a modbus sim (open ModSim) change to 16bit integer in menu, connect
2. create a simple write script
3. connect to sim 
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment**
 - FUXA version: Version 1.2.0-1815 
 - Node.js version: 18
 - npm version:
 - Platform/OS: windows 10
 - Browser: firefox

unocelli, 2024-09-19:
Hi, Thanks for the report

unocelli, 2024-09-20:
This issue should be fixed in the latest master branch version. let me know if you have any issues.

timmpo, 2024-09-20:
i did a run in VS. code and no error but it never runs the script when using write scipt. 
it newer triggers the if (scriptId && runtime !== undefined) (device-utils.js) runtime is undefined it just returning the input.
I have tried to digg deeper but my skill level is verry poor.

```
const callScaleScript = async (scriptId, params, runtime, isRead, value) => {
    if (scriptId && runtime !== undefined) {	// <--- When i test a write script, this if newer trigger (runtime is undefined)
        let parameters = [
            { name: 'value', type: 'value', value: value }
        ];
        let tagParams = [];
        if (params) {
            try {
                tagParams = JSON.parse(params);
            } catch (error) {
                runtime.logger.error(`'${tag.name}' error decoding ${isRead ? 'read' : 'write' } scale script params ${error.toString()}`);
            }
            parameters = [...parameters, ...tagParams];
        }
        const script = {
            id: scriptId,
            name: null,
            parameters: parameters,
            notLog: true
        };
        try {
            value = await runtime.scriptsMgr.runScript(script);
        } catch (error) {
            runtime.logger.error(`'${tag.name}' ${isRead ? 'read' : 'write'} script error! ${error.toString()}`);
        }
        return value;
    }
    return value;	// returning the input
}
```

timmpo, 2024-09-24:
I have now tested it again and it works perfect!
Testing Conditions: ModbusTCP
FUXA Version 1.2.0-1909

unocelli, 2024-10-27:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-10-27 by unocelli