frangoteam/FUXA #1372: [BUG] Unexpected Undo Behavior in Editor Crashing Projects

mauroalexandre, 2024-09-09:
I've encountered a peculiar issue with the 'undo' command (Ctrl + Z) in the editor. While I have yet to fully debug the problem, this issue has caused a complete crash of one of my projects, necessitating a restoration from backup. During this event, I observed that the 'svgcontent' tag in the JSON was empty, although all gadgets were correctly populated in the structure. I am including a brief video that showcases part of the problem. Notably, when using Ctrl+Z, the function starts malfunctioning, incorrectly restoring multiple elements simultaneously.

https://github.com/user-attachments/assets/ba7a8919-33dd-4fe0-a0a2-9160d951c717

unocelli, 2024-09-19:
Hi, Thanks for the report, I will try to fix it as soon as possible

unocelli, 2024-12-11:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-12-11 by unocelli