frangoteam/FUXA #1373: [FEATURE] add trend to output value control

GitTurboy, 2024-09-11:
**Describe the feature**
Add trend to output value control, when the user click on the control, a popup windows show the trend of the tag bind to it

**Describe the solution you'd like**
1. add a property to the output value control, such as "Show History trend", and how long of the  history period, if the user choose a tag has history data saved.
2. In the view interface, If the user  click the control, a popup window with trend will shown

**Additional context**

henjoe, 2024-09-14:
Done similar things to my other projects where it requires to show pop-up of Trends for a specific parameters. 

I've customized a bit of the existing Trend Chart to add "Series Mapping" where you can map or bind tags on OpenCard action. 
This is actually good features and highly useable on the industry. 

[image]

However it requires a lot of refactoring, I cannot PR for this due to the fact that it will affect a lot of the code in the existing fuxa (and it's kinda special cases, as far as the fuxa community concern).

unocelli, 2024-09-24:
#1256 #859