frangoteam/FUXA #1375: Write to Uint32MLE holding register

eyadalhanafi, 2024-09-12:
Hi, 

I'm trying to write to holding register type: Uint32MLE, but I got an issue:

_2024-09-11T10:18:35.220Z [error] 'UWP30' _writeMemory error! Error: Modbus exception 3: Illegal data value (value cannot be written to this register)_ 

I'm able to read the value from the register without any issue.

We are using Fuxa Version 1.1.17-1399

Regards,

rvbatista, 2024-09-16:
This message cames from your device, it's no accepting the value writed.

eyadalhanafi, 2024-09-17:
I'm writing to HR setting value in the controller UWP3.0 from Carlo Gavazzi. We will investigate that and check if the register is writable. we didn't get a confirmation from the manufacturer yet.

eyadalhanafi, 2024-09-18:
The issue comes from the device. You can close the case.

unocelli, 2024-10-27:
Let's close here, since there is nothing feasible

closed 2024-10-27 by unocelli