frangoteam/FUXA #1378: How to pass a variable parameter when connecting through a web API?

thions, 2024-09-13:
We have a device monitoring scenario where there are multiple fan devices on the main monitoring screen. Clicking on each fan will display a pop-up window showing the device profile of the current fan.

[image]

[image]

So when we click on the fan in the main interface, we bind an event and pop up a window. The data of the device file in the window is intended to be obtained through the API. However, we found that when connecting to the WEB API, we can only enter one URL and there is no place to set parameters. We would like to inquire about how to pass the parameters of the device changes as variables when binding the WEB API

unocelli, 2024-09-13:
#650

henjoe, 2024-09-14:
By the way you can actually do this using script, it might require you do some nodejs programming but it is worth it. 

And on your click event simply simply select action to "Run Script", you script something like this, and accepts your API URL as parameter. 

```
const axios = require('axios');
//For Get Request
try {
    const response = await axios.get(url, {
      headers: headers,
      params: params
    });
    console.log('GET Response:', response.data);
    return response.data;
  } catch (error) {
    handleError(error);
  }

```

Your headers, parameters and URL might be a parameter on your script, or you can hard coded it directly on your script.

thions, 2024-09-18:
Thank you everyone, let's give it a try

Daysons, 2024-09-19:
> By the way you can actually do this using script, it might require you do some nodejs programming but it is worth it.
>
> And on your click event simply simply select action to "Run Script", you script [...]

> By the way you can actually do this using script, it might require you do some nodejs programming but it is worth it.
>
> And on your click event simply simply select action to "Run Script", you script [...]

May I ask how to import Node's require module? Looking forward to your reply

henjoe, 2024-09-19:
> May I ask how to import Node's require module? Looking forward to your reply

i actually tried it, but it fails. 
I thought we can easily import any module that is in package.json, but I am wrong.

unocelli, 2025-03-04:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2025-03-04 by unocelli