frangoteam/FUXA #1380: npm Install fails on odbc on 1.2.0-1812, Debian Buster rpi CM4 armhf

robbudge, 2024-09-13:
Clean Rpi CM4, Debian Buster, armhf

it would appear that no valid install package exists for armhf. npm install fails, I have also ran the odbc install script directly with a not supported error.

Could the odbc requirement be disabled for unsupported platforms to avoid installation errors. Windows installation is bad enough

MatthewReed303, 2024-09-13:
@robbudge just remove node odbc from the `server/package.json` and then try again. I can confirm everything works fine on RPI CM4, any reason why you are using Buster and 32bit? I would flash the CM4 with Bookworm and 64bit and you won't have any issues at all.

robbudge, 2024-09-13:
Ok, I will give that a try and confirm.
The Debian Buster is the default on the unit, it’s an Industrial unit that utilizes the CM4 internally.

I will check with the manufacturer about a bookworm image.

MatthewReed303, 2024-09-13:
@robbudge what is the industrial device? Touch screen? You can flash the normal RPI image and login via SSH and install drivers manually etc also install the latest docker release. See the Wiki https://github.com/frangoteam/FUXA/wiki/Installing-and-Running

robbudge, 2024-09-13:
We have 2 under test a simple Din-Rail unit
<https://www.seeedstudio.com/EdgeBox-RPi-200-CM4102008-With-UPS-p-5729.html>
[image_7.jpeg]
EdgeBox RPi 200 - Raspberry Pi IoT Edge Device | IoT Gateway | 2GB RAM, 8GB eMMC, UPS, WiFi<https://www.seeedstudio.com/EdgeBox-RPi-200-CM4102008-With-UPS-p-5729.html>
seeedstudio.com<https://www.seeedstudio.com/EdgeBox-RPi-200-CM4102008-With-UPS-p-5729.html>

And a 10.1” CM4 screen.
<https://www.seeedstudio.com/reTerminal-DM-CM4104032-p-5898.html>
[sku114070262.jpg]
reTerminal DM - 10.1'' Industrial HMI | CM4104032 | Raspberry Pi Industrial Panel PC | Panel Controller<https://www.seeedstudio.com/reTerminal-DM-CM4104032-p-5898.html>
seeedstudio.com<https://www.seeedstudio.com/reTerminal-DM-CM4104032-p-5898.html>

I am always cautious of starting from scratch due to required hardware overlays needed.
They both work really well especially for the cost.

I did consider docker but adding out our images is a pain
I need to look how easy it would be to include ‘FileBrowser’ within the docker compose.

Sent from IPhone

MatthewReed303, 2024-09-14:
@robbudge I have used the edgebox 200 quite a few times, I con confirm standard RPI image is fine for that. I also add RT Kernel as I run CodeSys on them and work well. As for the touch screen everything you need is here https://wiki.seeedstudio.com/reterminal-dm-flash-OS/

As for file browser you can install Cockpit and add the file browser package and you have web ui for the Linux system. 

Note if you don't use Docker Image you will have to manually install the odbc drivers your self ( can try the install script )

robbudge, 2024-09-14:
It’s certainly a buster issue with odbc.
I am going to look into at upgrading the os.
The docker compose fails in buster also for 1.2.0-1812.

I think it’s just the edge box has an old OS as standard.
I was trying make a simple workflow for our guys without dismantling the edge box to perform the OS upgrade.

I like docker but updating _Images and customization becomes a lot harder.

Perhaps it would be beneficial to add these directories to the docker run command in the read.me guide to make them host folders ?
That way a docker could be used as standard, with user images & customization would be easily accessible.

MatthewReed303, 2024-09-14:
@robbudge see the Wiki install for more complete install options and building your own docker image https://github.com/frangoteam/FUXA/wiki/Installing-and-Running 

You should also be able to do an OS upgrade without flashing the CM4, just don't think you can change to 64bit this way. 

With the docker compose I use ./ which will store all the files in the same directory as the compose file, so everything is there in a single place.

robbudge, 2024-09-18:
ok new update,
just flashed the CM4 to bookworm.
arm64, Debian12.

sudo sh install_odbc_drivers.sh
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Hit:4 https://repos.influxdata.com/debian stable InRelease
Hit:5 https://packages.microsoft.com/debian/12/prod bookworm InRelease
Hit:6 http://archive.raspberrypi.com/debian bookworm InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
curl is already the newest version (7.88.1-10+deb12u7).
wget is already the newest version (1.21.3-1+b1).
unixodbc is already the newest version (2.3.11-2+deb12u1).
unixodbc-dev is already the newest version (2.3.11-2+deb12u1).
odbc-postgresql is already the newest version (1:13.02.0000-2+b1).
odbc-mariadb is already the newest version (3.1.15-3).
libsqliteodbc is already the newest version (0.9998-3+b1).
freetds-dev is already the newest version (1.3.17+ds-2).
freetds-bin is already the newest version (1.3.17+ds-2).
tdsodbc is already the newest version (1.3.17+ds-2).
libssl3 is already the newest version (3.0.14-1~deb12u2+rpt1).
libssl-dev is already the newest version (3.0.14-1~deb12u2+rpt1).
apt-transport-https is already the newest version (2.6.1).
The following packages were automatically installed and are no longer required:
  chromium-browser chromium-browser-l10n chromium-codecs-ffmpeg-extra
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
File '/usr/share/keyrings/microsoft-prod.gpg' exists. Overwrite? (y/N) y
install_odbc_drivers.sh: 69: [: arm64: unexpected operator
install_odbc_drivers.sh: 69: [: arm64: unexpected operator
install_odbc_drivers.sh: 80: [: arm64: unexpected operator
MS SQL ODBC driver installation skipped due to unsupported architecture or Debian version.
Architecture: arm64
Debian Version: 12
Check if /usr/lib/odbc/ directory exists, if not, create it
Move ODBC driver files based on system architecture
install_odbc_drivers.sh: 105: [: arm64: unexpected operator
install_odbc_drivers.sh: 108: [: arm64: unexpected operator
install_odbc_drivers.sh: 111: [: arm64: unexpected operator
install_odbc_drivers.sh: 111: [: arm64: unexpected operator
Move ODBC driver files Unsupported architecture: arm64

ODBC still has issues.
Going to play around and see if docker works

MatthewReed303, 2024-09-18:
@robbudge try the following

`sudo apt-get update && sudo apt-get install -y dos2unix`

`sudo apt-get install -y build-essential unixodbc unixodbc-dev`

`sudo dos2unix install_odbc_drivers.sh && sudo chmod +x install_odbc_drivers.sh`

It may be an format error with the script, as I just tested the script on my end and works fine on RPI arm64 with bookworm. 

If you still don't have any luck, just stick with the Docker Version.

robbudge, 2024-09-18:
@MatthewReed303 
Same, strange that the script if failing first at line 69, unexpected operator

File '/usr/share/keyrings/microsoft-prod.gpg' exists. Overwrite? (y/N) y
install_odbc_drivers.sh: 69: [: arm64: unexpected operator
install_odbc_drivers.sh: 69: [: arm64: unexpected operator
install_odbc_drivers.sh: 80: [: arm64: unexpected operator
MS SQL ODBC driver installation skipped due to unsupported architecture or Debian version.
Architecture: arm64
Debian Version: 12
Check if /usr/lib/odbc/ directory exists, if not, create it
Move ODBC driver files based on system architecture
install_odbc_drivers.sh: 105: [: arm64: unexpected operator
install_odbc_drivers.sh: 108: [: arm64: unexpected operator
install_odbc_drivers.sh: 111: [: arm64: unexpected operator
install_odbc_drivers.sh: 111: [: arm64: unexpected operator

cnnews, 2024-11-04:
sudo apt install build-essential

cnnews, 2024-11-07:
```
npm ERR! code 1
npm ERR! path /usr/local/lib/node_modules/@frangoteam/fuxa/node_modules/odbc
npm ERR! command failed
npm ERR! command sh -c node-pre-gyp install --fallback-to-build
npm ERR! make: Entering directory '/usr/local/lib/node_modules/@frangoteam/fuxa/       node_modules/odbc/build'
npm ERR!   CXX(target) Release/obj.target/odbc/src/odbc.o
npm ERR! make: Leaving directory '/usr/local/lib/node_modules/@frangoteam/fuxa/n       ode_modules/odbc/build'
npm ERR! Failed to execute '/usr/bin/node /usr/share/nodejs/node-gyp/bin/node-gy       p.js build --fallback-to-build --module=/usr/local/lib/node_modules/@frangoteam/       fuxa/node_modules/odbc/lib/bindings/napi-v8/odbc.node --module_name=odbc --modul       e_path=/usr/local/lib/node_modules/@frangoteam/fuxa/node_modules/odbc/lib/bindin       gs/napi-v8 --napi_version=9 --node_abi_napi=napi --napi_build_version=8 --node_n       api_label=napi-v8' (1)
npm ERR! node-pre-gyp info it worked if it ends with ok
npm ERR! node-pre-gyp info using node-pre-gyp@1.0.11
npm ERR! node-pre-gyp info using node@18.19.0 | linux | arm
npm ERR! node-pre-gyp info check checked for "/usr/local/lib/node_modules/@frang       oteam/fuxa/node_modules/odbc/lib/bindings/napi-v8/odbc.node" (not found)
npm ERR! node-pre-gyp http GET https://github.com/IBM/node-odbc/releases/downloa       d/v2.4.9/odbc-v2.4.9-linux-arm-napi-v8.tar.gz
npm ERR! node-pre-gyp ERR! install response status 404 Not Found on https://gith       ub.com/IBM/node-odbc/releases/download/v2.4.9/odbc-v2.4.9-linux-arm-napi-v8.tar.       gz
npm ERR! node-pre-gyp WARN Pre-built binaries not installable for odbc@2.4.9 and        node@18.19.0 (node-v108 ABI, glibc) (falling back to source compile with node-g       yp)
npm ERR! node-pre-gyp WARN Hit error response status 404 Not Found on https://gi       thub.com/IBM/node-odbc/releases/download/v2.4.9/odbc-v2.4.9-linux-arm-napi-v8.ta       r.gz
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@9.3.0
npm ERR! gyp info using node@18.19.0 | linux | arm
npm ERR! gyp info ok
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@9.3.0
npm ERR! gyp info using node@18.19.0 | linux | arm
npm ERR! gyp info find Python using Python version 3.11.2 found at "/usr/bin/pyt       hon3"
npm ERR! gyp info spawn /usr/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   '/usr/share/nodejs/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'make',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/usr/local/lib/node_modules/@frangoteam/fuxa/nod       e_modules/odbc/build/config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/usr/share/nodejs/node-gyp/addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/usr/include/nodejs/common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=/usr/include/nodejs',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=/usr/share/nodejs/node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=/usr/include/nodejs/<(target_arc       h)/node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=/usr/local/lib/node_modules/@f       rangoteam/fuxa/node_modules/odbc',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info ok
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@9.3.0
npm ERR! gyp info using node@18.19.0 | linux | arm
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! In file included from ../src/odbc.cpp:22:
npm ERR! ../src/odbc.h:37:10: fatal error: sql.h: No such file or directory
npm ERR!    37 | #include <sql.h>
npm ERR!       |          ^~~~~~~
npm ERR! compilation terminated.
npm ERR! make: *** [odbc.target.mk:125: Release/obj.target/odbc/src/odbc.o] Erro       r 1
npm ERR! gyp ERR! build error
npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
npm ERR! gyp ERR! stack     at ChildProcess.onExit (/usr/share/nodejs/node-gyp/l       ib/build.js:203:23)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:517:28)
npm ERR! gyp ERR! stack     at ChildProcess._handle.onexit (node:internal/child_       process:292:12)
npm ERR! gyp ERR! System Linux 6.6.54-current-sunxi
npm ERR! gyp ERR! command "/usr/bin/node" "/usr/share/nodejs/node-gyp/bin/node-g       yp.js" "build" "--fallback-to-build" "--module=/usr/local/lib/node_modules/@fran       goteam/fuxa/node_modules/odbc/lib/bindings/napi-v8/odbc.node" "--module_name=odb       c" "--module_path=/usr/local/lib/node_modules/@frangoteam/fuxa/node_modules/odbc       /lib/bindings/napi-v8" "--napi_version=9" "--node_abi_napi=napi" "--napi_build_v       ersion=8" "--node_napi_label=napi-v8"
npm ERR! gyp ERR! cwd /usr/local/lib/node_modules/@frangoteam/fuxa/node_modules/       odbc
npm ERR! gyp ERR! node -v v18.19.0
npm ERR! gyp ERR! node-gyp -v v9.3.0
npm ERR! gyp ERR! not ok
npm ERR! node-pre-gyp ERR! build error
[... cut ...]
npm ERR! node-pre-gyp ERR! cwd /usr/local/lib/node_modules/@frangoteam/fuxa/node       _modules/odbc
npm ERR! node-pre-gyp ERR! node -v v18.19.0
npm ERR! node-pre-gyp ERR! node-pre-gyp -v v1.0.11
npm ERR! node-pre-gyp ERR! not ok

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2024-11-06T23_30_12_657Z-debug-0.log
```
This is my npm install log，and this is the most important
**npm ERR! ../src/odbc.h:37:10: fatal error: sql.h: No such file or directory**
so，I install the unixodbc-devel pack

unocelli, 2025-03-04:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2025-03-04 by unocelli