frangoteam/FUXA #1391: I saw your program code on another code hosting platform being used for charging users.

RilesLee, 2024-09-17:
I saw the source code of your program on another platform, and it is being used to charge users. I'm not sure if your program code is open for commercial use or if it allows redistribution, etc. However, I believe that copying open-source code and charging for it is wrong. If you want to protect your developer rights, please reply to me so we can discuss this further.

henjoe, 2024-09-18:
Just read the open license bro, it was already indicated there. 
That's the core concept of opensource, anyone can use it regardless commercial or personal.

unocelli, 2024-09-27:
However, the MIT licence does not prevent you from contributing to the project 😉

closed 2024-09-27 by unocelli