frangoteam/FUXA #1400: Does fuxa-editor support generating icons using code？How?

Daysons, 2024-09-20:


unocelli, 2024-09-20:
Please give an example

Daysons, 2024-09-23:
> Please give an example

Our current functionality allows us to select an icon on the left side and generate it on the right canvas by clicking. I want to implement this process using code. I've managed to select the icon on the left, but now I want to place the icon on the canvas using code instead of mouse clicks.
[image]

unocelli, 2024-09-29:
if you intend to execute a script when the button is clicked, you must write a script and then associate the script with the click

Daysons, 2024-09-30:
> if you intend to execute a script when the button is clicked, you must write a script and then associate the script with the click

i

> if you intend to execute a script when the button is clicked, you must write a script and then associate the script with the click

> if you intend to execute a script when the button is clicked, you must write a script and then associate the script with the click

The documentation related to click events is the source code of the https://github.com/frangoteam/svgedit repository?

unocelli, 2024-10-02:
No, the handler of button click is in FUXA frontend (client)

unocelli, 2024-10-27:
Let's close here, since there is nothing feasible

closed 2024-10-27 by unocelli