frangoteam/FUXA #1401: [BUG] Add Widget Error

henjoe, 2024-09-20:
**Describe the bug**
I am not actually if this is BUG or Not, but I was just trying the add widget features, and use the 
```
dynamicIndicatorTestButtonSVG-init-edit.svg
```
as the test svg file, When I imported it, it gives me this error:
```
 [webpack-dev-server] [HPM] Error occurred while proxying request localhost:4200/resources/dynamicIndicatorTestButtonSVG-init-edit.svg to http://localhost:1889/ [ECONNREFUSED] (https://nodejs.org/api/errors.html#errors_common_system_errors)
```

**To Reproduce**
Steps to reproduce the behavior:
1. Add Image Link control and look for the sample SVG.

**Expected behavior**
Added new widget. 

**Screenshots**
[image]

**Environment**
 - FUXA version: just the master branch
 - Node.js version: V18.xx
 - npm version: 10.2.3
 - Platform/OS: Windows
 - Browser: Chrome

unocelli, 2024-09-20:
Hi, try to import it via 'Image tool (embedded)' 
[image]

henjoe, 2024-09-20:
Using the Image Tool (embedded) 

[image]

unocelli, 2024-09-20:
Hi, I don't know what you are trying 😉 
[image]

henjoe, 2024-09-20:
Does this works when in debug mode (ng serve on the client)?  Or the client should be build first? 
I am not quite sure whats wrong in here, but I might need to check more.

unocelli, 2024-09-20:
should work in debug mode too

unocelli, 2024-10-27:
Let's close here, since there is nothing feasible

closed 2024-10-27 by unocelli