frangoteam/FUXA #1403: Auto-Delete historical tags from database in FIFO order

Kiruthick-Roxx, 2024-09-20:
Describe the feature

It will be good to to set the Auto-Delete feature for individual tags that has been enabled with registration since the amount of data that it will consume if it's going to more than 5000 tags will be huge. One more option to set the threshold value to register tags based on change can be added so that it will record only if it's above or below the threshold as even decimal points are detected as change and will be recorded 

Describe the solution you'd like

An option can be added in the Tag options property itself to determine how long the data should retain for individual tags and will be deleted in First In and First Out (FIFO) order. Also a block to enter the threshold value for that particular tag to register data during change, can be given.

henjoe, 2024-09-20:
This is what we call as "deadband", it will only historized based on the amount of "change". 
As per my experience this is important when having issues on the database size, but as the technologies advanced, a lot of database has advance algorithm to compress the size of its size, so over the years this "deadband" features is not necessarily anymore (at least on my opinion). 

For the FIFO, you can do this outside the FUXA, you can set data retention on the database itself.

unocelli, 2024-09-20:
Hi, I think too that the FIFO in FUXA could compromise the performance

Kiruthick-Roxx, 2024-09-21:
Hi,
Even though the database compresses the data, will it be effective when the system records more than 10000 tags for a period of 1 year or 6 months. Also for the database setup, can you please make a video on how to setup and connect it FUXA via ODBC as a video demonstration would be much more helpful.

unocelli, 2024-09-29:
if you intend to run a script on the click of the button, you must write a script and then associate the script with the click

Kiruthick-Roxx, 2024-10-03:
Hi,
 I'm yet to setup ODBC. I'll try and put up any questions if I have.
 
 Thanks

unocelli, 2024-10-27:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-10-27 by unocelli