frangoteam/FUXA #1406: [BUG]Bar chart is showing total kWh consumption

jlumalang02, 2024-09-25:
Im having a problem with showing data using bar chart. Are they anyway i can group the consumption of kWh to daily or monthly consumption? 

[image]

unocelli, 2024-09-29:
Yes, the tag that you have bind to chart is in KW? please provide how you have configured your bar chart.

jlumalang02, 2024-09-30:
[image]

[image]

Here's the configuration of the bar chart. Im pretty new here on fuxa and have little experience in SCADA. I hope you can help with this. 

The tag is in kWh and i need to get the daily consumption, monthly consumption and yearly consumption. 

Fuxa version: V1.2.00
Running on windows 11

unocelli, 2024-10-02:
Hi, we have to fix and extend the options

jlumalang02, 2024-10-03:
hi already updated the fuxa application however only hour sum still showing and the data is still cumulative but grouped as per hour.

unocelli, 2024-10-03:
In the branch master you can try the version with the fix, please let me know if the calculation is correct.

unocelli, 2024-10-27:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-10-27 by unocelli