frangoteam/FUXA #1408: [FEATURE] Retain the last value added into the input value

prac05, 2024-09-25:
**Describe the feature**
My colleague is using an input value controller to store the desired value of a parameter. He says that after several reboots, these values have been lost. Is there any feature that prevents data loss? Does the DAQ storage configuration influence this? Also he is using Dialog Mode of input field.

**Additional context**
[image]
[image]

MatthewReed303, 2024-09-25:
@prac05 you should be using retain variables on the PLC and not with Fuxa. If the variable is changed on the PLC it's stored within the PLC

prac05, 2024-09-25:
The variable is already being stored in the PLC via Node-red.  It is not practical that when the value inserted in the input is lost, it has to be looked for elsewhere.  So is there some way to always see the last value inserted in case the PLC suffers a power failure?

MatthewReed303, 2024-09-25:
@prac05 that's exactly what retain value does in the PLC, if power is cycled it's still retained? What PLC are you using? On the Input make sure you have the updates enabled 

[image]

prac05, 2024-09-26:
Hi Matthew,

On the input settings, what is the function of the value parameter?

[image]

unocelli, 2024-09-29:
Hi, you can use enable the restore option on the Tag

unocelli, 2024-10-27:
Let's close here, since there is nothing feasible

closed 2024-10-27 by unocelli

andy8601, 2024-12-16:
> 嗨，您可以在标签上启用恢复选项

“to restore” This does not take effect for external DAQ storage。 InfluxDB