frangoteam/FUXA #1411: [BUG] some objects disappear with no-admin users

zblack88, 2024-09-26:
**Describe the bug**
If you enable the authentication with token, when you are logged with a non-admin account some objects (lines, rectangles, etc.) disappear, even if no visibility has been set on them.
It seems random, because on identical pages different objects disappear.

With admin account everything works fine.

Any idea of 

**Environment**
 - FUXA version: 1.2.0
 - npm version: 8.11.0
 - Platform/OS: Windows 10
 - Browser: Edge and Chrome

unocelli, 2024-09-29:
Hi, please provide a example.

zblack88, 2024-09-30:
This is the output:
admin account:
[image]

view account (all privileges but "Administrator"):
[image]

No authorization has beeen set for any object:
[image]

If you want I can share the .json project in pvt

unocelli, 2024-10-02:
Yes, please send me your project [4frango@gmail.com](mailto:4frango@gmail.com)

unocelli, 2024-10-06:
Hi, This issue should be fixed in the latest master branch version. let me know if you have any issues.

zblack88, 2024-10-06:
Thank you very much, it works now.

unocelli, 2024-10-27:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-10-27 by unocelli