frangoteam/FUXA #1420: [BUG]

zblack88, 2024-10-03:
**Describe the bug**
When trying to create any script, I receive an "Unauthorized" error, but I have no authorization set, neither for the script, nor for the project.

>Http failure response for http://localhost:1881/api/runscript: 401 Unauthorized
>Unauthorized!

**To Reproduce**
Hard to say, on another machine with the same fuxa version worked correctly.

**Expected behavior**
write "ciao" on console

**Screenshots**
[image]

**Environment**
 - FUXA version: 1.2.0-1922
 - Node.js version: v18.20.4 
 - npm version:
 - Platform/OS: Windows 10
 - Browser: Edge and Chrome

unocelli, 2024-10-03:
Hi, check your installation or system
[image]

adnanebrahimi, 2024-10-05:
I had the same issue. To fix it, I removed the Docker volumes of Fuxa on the host and recreated the container. The issue has been resolved.

Here is my docker compose service of Fuxa:

 ```yml
 fuxa:
    image: frangoteam/fuxa:latest
    container_name: fuxa
    restart: always
    ports:
      - "1881:1881" 
    volumes:
      - './fuxa_volume/appdata:/usr/src/app/FUXA/server/_appdata'
      - './fuxa_volume/db:/usr/src/app/FUXA/server/_db'
      - './fuxa_volume/logs:/usr/src/app/FUXA/server/_logs'
      - './fuxa_volume/images:/usr/src/app/FUXA/server/_images' 
    depends_on:
      - influxdb
    networks:
      - iiot_network
```

RikvdReijen, 2024-10-13:
Hey guys, I noticed when I started a new project and executed the script it works. However on entering a connection I start to get the error.
What I did notice is that when I first save the script (by pressing ok in the bottom right) and then re-execute the script it works. I am gonna monitor if the problem reoccurs and let you guys know.

lewsut, 2024-10-13:
I also get this often too

RikvdReijen, 2024-10-13:
Hey Lewsut! Thanks for your quick response! It turned out to be exactly that! Even if the error is in a totally different script it will still break the other scripts.

For anyone wondering how you can set the value of a button to zero again with delay after clicking, this is how I did it, took me a day to figure out haha;
$setTag(pTag, 1);
setTimeout(function() {
    // Code to be executed after the delay
    // For example, updating a tag value:
    $setTag(pTag, 0);
}, ResetDelayMilliseconds);

unocelli, 2024-10-27:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-10-27 by unocelli