frangoteam/FUXA #1433: [FEATURE] Add Calendar Control

henjoe, 2024-10-10:
**Describe the feature**
Calendar control is good for setting up values related to Date.
[image]

**Describe the solution you'd like**
Use any angular-calendar component for this.

** Others ** 
Also some suggestions, I know a lot of people here knows how to code in Angular, it would be much better if you have some tutorials on how to add them as control in Fuxa, that way you are helping to create additional control components in fuxa by community, just for example, I can code related to this angular calendar, however I don't know how to integrate them with fuxa, if you have some tutorial to do it, it would be much easier for the community to PR for the new control they want.

MatthewReed303, 2024-10-10:
@henjoe  you could create this as a Widget... I'm pretty sure you can use HTML tags in the SVG along with the Javascript tags, I will build some Time adjusting and display widgets soon for a project I'm working on.
https://github.com/frangoteam/FUXA/wiki/HowTo-Widgets

henjoe, 2024-10-10:
> @henjoe  you could create this as a Widget... I'm pretty sure you can use HTML tags in the SVG along with the Javascript tags, I will build some Time adjusting and display widgets soon for a project I'm working on.
> https://github.com/frangoteam/FUXA/wiki/HowTo-Widgets

Hi thanks for the tip! One thing I am curious what application software you use for creating customized widgets/SVG?