frangoteam/FUXA #1435: [BUG] ScaleRead Script Return Null when value is Zero (0)

henjoe, 2024-10-12:
**Describe the bug**
When the return value from the script is Zero, the value will be 
```
Script OK: null
```

**To Reproduce**
Steps to reproduce the behavior:
1. Add any Tag on any device
2. Create script that will always return Zero value for example: 
```
return value * 0
```
4. See the return value in the Device list page.

**Expected behavior**
It should return Zero value.

**Screenshots**
[image]

**Environment**
 - FUXA version: latest 1.2.0
 - Node.js version:  v18.X
 - npm version:
 - Platform/OS:
 - Browser:

henjoe, 2024-10-12:
Under the Server\runtime\scripts\index.js
Simply edit the line 84 from
```
resolve(result || `Script OK: ${script.name}`);
```
to this, solve the problem.
```
resolve(result);
```

Just simply remove the ``` || `Script OK: ${script.name}```` solves the problem. 
I don't what is this for, but yeah.

unocelli, 2024-10-27:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-10-27 by unocelli