frangoteam/FUXA #1436: [BUG], ODBC Not supported in all platforms

robbudge, 2024-10-13:
**Describe the bug**
I am trying to deploy via docker on a wago 752-8303 edge controller.

Docker build fails as no package can be found for the wago linux build, this a closed system with docker support only, ssh is available but as far as I can determine no package or installation manager only docker.
This processor would be an ideal partner for fuxa 

Sending build context to Docker daemon   27.4MB
Step 1/11 : FROM node:14
14: Pulling from library/node
166ef90dbaea: Pull complete 
84460c1688b7: Pull complete 
0ed477965486: Pull complete 
a6bfe04b18e6: Pull complete 
fc69a47db99b: Pull complete 
c717d23dd91b: Pull complete 
2fb3a51bf87c: Pull complete 
c216a8fe7d76: Pull complete 
41073937a20a: Pull complete 
Digest: sha256:a158d3b9b4e3fa813fa6c8c590b8f0a860e015ad4e59bbce5744d2f6fd8461aa
Status: Downloaded newer image for node:14
 ---> 99d8b9f1b49d
Step 2/11 : WORKDIR /usr/src/app
 ---> Running in da9c6ee03b0e
 ---> Removed intermediate container da9c6ee03b0e
 ---> 72f8717c835f
Step 3/11 : RUN git clone https://github.com/frangoteam/FUXA.git
 ---> Running in deeb19c3e383
Cloning into 'FUXA'...
Checking out files: 100% (1362/1362), done.
 ---> Removed intermediate container deeb19c3e383
 ---> 35b63bb00f63
Step 4/11 : WORKDIR /usr/src/app/FUXA
 ---> Running in ba093614b37e
 ---> Removed intermediate container ba093614b37e
 ---> 75adc62098cb
Step 5/11 : WORKDIR /usr/src/app/FUXA/server
 ---> Running in d59f8dbd09f2
 ---> Removed intermediate container d59f8dbd09f2
 ---> a81b81766fbe
Step 6/11 : RUN npm install
 ---> Running in 6ace97f61852

> @serialport/bindings-cpp@12.0.1 install /usr/src/app/FUXA/server/node_modules/@serialport/bindings-cpp
> node-gyp-build

> odbc@2.4.9 install /usr/src/app/FUXA/server/node_modules/odbc
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! install response status 404 Not Found on https://github.com/IBM/node-odbc/releases/download/v2.4.9/odbc-v2.4.9-linux-arm-napi-v8.tar.gz 
node-pre-gyp WARN Pre-built binaries not installable for odbc@2.4.9 and node@14.21.3 (node-v83 ABI, glibc) (falling back to source compile with node-gyp) 
node-pre-gyp WARN Hit error response status 404 Not Found on https://github.com/IBM/node-odbc/releases/download/v2.4.9/odbc-v2.4.9-linux-arm-napi-v8.tar.gz 
make: Entering directory '/usr/src/app/FUXA/server/node_modules/odbc/build'
  CXX(target) Release/obj.target/odbc/src/odbc.o
In file included from ../src/odbc.cpp:22:
../src/odbc.h:37:10: fatal error: sql.h: No such file or directory
 #include <sql.h>
          ^~~~~~~
compilation terminated.
make: Leaving directory '/usr/src/app/FUXA/server/node_modules/odbc/build'
make: *** [odbc.target.mk:125: Release/obj.target/odbc/src/odbc.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:400:28)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:285:12)
gyp ERR! System Linux 5.15.107-rt62-w04.03.05
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/usr/src/app/FUXA/server/node_modules/odbc/lib/bindings/napi-v8/odbc.node" "--module_name=odbc" "--module_path=/usr/src/app/FUXA/server/node_modules/odbc/lib/bindings/napi-v8" "--napi_version=8" "--node_abi_napi=napi" "--napi_build_version=8" "--node_napi_label=napi-v8"
gyp ERR! cwd /usr/src/app/FUXA/server/node_modules/odbc
gyp ERR! node -v v14.21.3
gyp ERR! node-gyp -v v5.1.1
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/src/app/FUXA/server/node_modules/odbc/lib/bindings/napi-v8/odbc.node --module_name=odbc --module_path=/usr/src/app/FUXA/server/node_modules/odbc/lib/bindings/napi-v8 --napi_version=8 --node_abi_napi=napi --napi_build_version=8 --node_napi_label=napi-v8' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/usr/src/app/FUXA/server/node_modules/@mapbox/node-pre-gyp/lib/util/compile.js:89:23)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:400:28)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:1088:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:296:5)
node-pre-gyp ERR! System Linux 5.15.107-rt62-w04.03.05
node-pre-gyp ERR! command "/usr/local/bin/node" "/usr/src/app/FUXA/server/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /usr/src/app/FUXA/server/node_modules/odbc
node-pre-gyp ERR! node -v v14.21.3
node-pre-gyp ERR! node-pre-gyp -v v1.0.11
node-pre-gyp ERR! not ok 
Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/src/app/FUXA/server/node_modules/odbc/lib/bindings/napi-v8/odbc.node --module_name=odbc --module_path=/usr/src/app/FUXA/server/node_modules/odbc/lib/bindings/napi-v8 --napi_version=8 --node_abi_napi=napi --napi_build_version=8 --node_napi_label=napi-v8' (1)
npm WARN notsup Unsupported engine for odbc@2.4.9: wanted: {"node":">=18.0.0"} (current: {"node":"14.21.3","npm":"6.14.18"})
npm WARN notsup Not compatible with your version of node/npm: odbc@2.4.9
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@~2.3.2 (node_modules/chokidar/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.3.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! odbc@2.4.9 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the odbc@2.4.9 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2024-10-13T01_07_22_428Z-debug.log
The command '/bin/sh -c npm install' returned a non-zero code: 1

As ODBC is not supported on all platforms can it be selectable as a plug-in or be omitted from the build sequence when a unsupported platform is detected.

MatthewReed303, 2024-10-13:
@robbudge any reason why you are trying to build and not just using the prebuilt images? 

https://github.com/frangoteam/FUXA/wiki/Installing-and-Running

Wago is pretty locked down, but I did use Fuxa on one a while ago via Docker no issue, try with prebuilt image first, you can access the OS via SSH or use CodeSys to transfer files also.

robbudge, 2024-10-13:
This hardware hard the earlier non ODBC version installed previously without issues. 
I was attempting to install the latest on my Wago sandbox 
In will attempt with a docker pull not build.

unocelli, 2024-10-27:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2024-10-27 by unocelli