frangoteam/FUXA #1440: [BUG] Data doesn't refresh

Kapket, 2024-10-14:
**Describe the bug**
We have a problem because in the connections we retrieve data through the API and the columns with the value and timestamp don't refresh. For this reason, the alarms don't want to work for us.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to connections
2. Add API link to data
3. Go to list view
4. Data under value and timestamp won't refresh

**Expected behavior**
I expect that the data in the connections will refresh.

**Environment**
- FUXA version: V.1.1.18-1460
- Node.js version: v22.06.0 
- npm version: 10.8.2
- Platform/OS: Ubuntu 22.04.4 LTS
- Browser: Chrmoe / Firefox / Brave

unocelli, 2024-10-16:
Hi, try nodejs version 18 (supported) and FUXA 1.2.0

unocelli, 2024-10-27:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2024-10-27 by unocelli