frangoteam/FUXA #1441: [BUG] Button Fails to Function Properly Without Displaying Variable on Screen

martinjug, 2024-10-14:
First recording is showing **Button Fails to Function Properly Without Displaying Variable on Screen**

https://github.com/user-attachments/assets/c7f109f2-2757-4124-8d34-f7b8c8d39085

**Environment**
 - FUXA version: **1.2.0-1956**
 - Node.js version:  **v14.21.3**
 - npm version: **6.14.18**
 - Platform/OS: **Windows 10**
 - Browser: **Brave**
 - Driver: **node-snap7": "1.0.6"**

Second recording is showing previous version, everything works as expected.

https://github.com/user-attachments/assets/5c60209b-33d1-49df-b209-30898b64de88

**Environment**
 - FUXA version: **1.1.19-1619**
 - Node.js version:  **v14.21.3**
 - npm version: **6.14.18**
 - Platform/OS: **Windows 10**
 - Browser: **Brave**
 - Driver: **node-snap7": "1.0.6"**

martinjug, 2024-10-16:
I realize my previous explanation might not have been clear. I've made another recording demonstrating the problem. On the left side of screen is Modbus server publishing one coil. On the right side of screen is FUXA.

"ToggleValue" button works as expected in version V1.19 but does not work in V1.20. Specifically, in V1.20, it fails to toggle value of a coil(you can see it's not changing on modbus server). **I think that script that toggles value is not executed in V1.20.**

When toggling the value directly in the Modbus server, both V1.19 and V1.20 reflect the change correctly. However, when using the toggle button in FUXA, it works as intended in V1.19 but fails in V1.20.

https://github.com/user-attachments/assets/d1bf527b-5344-4059-b580-f1946468018b

https://github.com/user-attachments/assets/fc52912a-7ba5-4347-824e-eae4209add43

unocelli, 2024-10-24:
[image]

martinjug, 2024-10-27:
Your example seems to be working as expected. If I'm able to reproduce the bug, I'll post the details here. It's definitely strange behavior.

unocelli, 2024-12-11:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2024-12-11 by unocelli