frangoteam/FUXA #1447: [BUG] Scripting doesn't work

Brandon-sistein, 2024-10-19:
I tried this script:

let test1 ="Hello world";

let test2 = "`SELECT * " + test1 + "`"
console.log(test2);

and show me this error:

```
>Http failure response for http://192.168.0.11:1881/api/runscript: 401 Unauthorized
>Unauthorized!
```

How can I solve it? and  why it is happening?

Environment
 - FUXA Docker
 -Raspberry

Brandon-sistein, 2024-10-19:
This is the sistem error:

[image]
[image]

robbudge, 2024-10-24:
Same Issue also 

Version 1.2.0-1956 
npm 10.7.0
node v18.20.4
SeeedStudio RPI-200 running buster

[image]
This was working fine, very recently and can only be related to the project configuration.
Only major change was to enable security.
I have since disabled security and still the same will keep looking

robbudge, 2024-10-24:
Quick Update, this is also noted on a previous post.
if you have any script with an error, it will cause this issue with all script executions.

I had a simple scratch script that I use getTag to obtain the tags uuid for grafana. Obviously this was a  script full of junk.
Once it was deleted all my other scripts worked fine.

Brandon-sistein, 2024-10-29:
I deleted all the scripts and create a "hello world" and still showing the error

unocelli, 2024-11-06:
Hi, I confirm that there are some problems if you leave code commented with // in scripts. we are investigating, for the time being we recommend not using // to comment code

WuBingo12, 2024-11-10:
If there is a comment at the last line of any script using // to comment code, this issue will occur

unocelli, 2024-11-10:
@WuBingo12 Thanks for the report, is exactly what I was looking for

unocelli, 2024-12-11:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-12-11 by unocelli