frangoteam/FUXA #1448: [BUG]为什么我将livedemo里的项目导入到最新版的fuxa里，但是他不工作了，是要设置哪里吗？

wen089130, 2024-10-21:
**Describe the bug**
A clear and concise description of what the bug is.
[image]

**To Reproduce**
Steps to reproduce the behavior:
[fuxa-project.json](https://github.com/user-attachments/files/17455877/fuxa-project.json)

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment**
 - FUXA version: 
 - Node.js version: 
 - npm version:
 - Platform/OS:
 - Browser:

unocelli, 2024-10-24:
Hi, You must activate Device A

unocelli, 2024-12-11:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2024-12-11 by unocelli