frangoteam/FUXA #1449: server connection failed

DHIAFIQ, 2024-10-21:
hi fuxa team, 

can I know, what the factor of causes the sever connection failed,
my polling right now is 1.5 sec but the error still occur.

this the picture of the error
[image]

ShahxlaN, 2024-10-22:
i had the same problem..the command prompt does not show any error, but suddenly the error occur

may i know the main factor of this error causes?
before this it never happen,
i already use the latest version but the error still occur.

unocelli, 2024-10-24:
Server send every 10 seconds ALIVE message to every connected client, the client that don't receive ALIVE after 20 seconds show this error. I would look to see if the server is blocked.

DHIAFIQ, 2024-11-04:
I run the fuxa in the same pc, so its no firewall between the server and the client

unocelli, 2024-11-06:
Try to investigate with chrome devtools

unocelli, 2024-12-11:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2024-12-11 by unocelli