frangoteam/FUXA #1457: [BUG] attachment corrupted when sending report by email

zblack88, 2024-10-25:
**Describe the bug**
When you set up and execute a report, if you download it from the editor interface it works, but if you try to open the report from the received email the attachment is corrupted.
I saw the dimension of the attachment is lower (2.5kB) then the original report dimension (15kB).

**To Reproduce**
Steps to reproduce the behavior:
1. Set up a  Report filling the "email recipient address" field.
2. Set up properly the SMTP service
3. Execute the report
4. See the received email

**Expected behavior**
Email attachment equal to the report downloaded from the editor interface

**Environment**
 - FUXA version: 1.1.20
 - Platform/OS: windows 10
 - Browser: chrome and edge

jingshui127, 2024-11-21:
不能正确显示中文字符。
比如变量是中文的，PDF将显示为乱码！

jingshui127, 2024-11-26:
[image]

unocelli, 2024-11-29:
Hi, is it possible that it is an OS problem? installed in docker container? Try investigating

unocelli, 2024-12-11:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2024-12-11 by unocelli

zblack88, 2024-12-11:
To be honest, it have not been resolved.
Checked everything, but attachment continue coming corrupted.
If you want I can forward the email received.

unocelli, 2025-02-14:
The problem is not to have e corrupted attachment, is how we can reproduce the issue