frangoteam/FUXA #1466: pipe colour

andy8601, 2024-10-30:
Should we add conditions for the pipeline content in the pipeline setting page, such as when a variable is 1, the pipeline content is blue, and when the variable is 0, the pipeline content is red? I tried to draw two identical pipeline coverage, and set the overall display or hide, in order to achieve such a functional effect, but it is not ideal, sometimes bugs will appear, and the two pipelines are displayed at the same time.
[image]

mauroalexandre, 2024-11-04:
It's a good feature suggestion. Ideally, a new Action should be created in the Pipe with the name "Change style."

When I have some free time, I'll try to implement it.

unocelli, 2024-12-03:
@mauroalexandre Hi, are you working on it?

unocelli, 2024-12-06:
I'm working on this feature

andy8601, 2024-12-22:
Didn't find an option for pipe color?
[image]
[image]
@ @unocelli 

> I'm working on this feature

unocelli, 2024-12-22:
[image]

andy8601, 2024-12-22:
Maybe I did not express it clearly. The color of the pipe I said is not blink, I think it is meaningless for the pipe to blink. What I want to express is that it does not affect the flow of the pipe, for example, when the temperature is 10°, the pipe content flows red, and when the temperature is 20°, the pipe content flows yellow. @unocelli

unocelli, 2024-12-22:
[image]

andy8601, 2024-12-25:
Yes, it currently works!

unocelli, 2025-01-01:
I’m going to close this as resolved. let me know if you have any issues.

closed 2025-01-01 by unocelli