frangoteam/FUXA #1467: [FEATURE] progress bar and cancel trend load

mariorl, 2024-10-30:
**Describe the feature**
If you sample fast, daq files become too huge, and need minutes for loading on front end, and eventually will not load at all.

**Describe the solution you'd like**
- Progress bar with cancel: if the query is too big you'll see it on the progress bar (will grow very slow) and cancel trend rendering if needed.
- A way for select a very concise querý, for example 1 single hour, will imply a quite smaller dataset. May be there's already a way for do this from scripting, but its not clear how to do it.

**Additional context**
I'm using a modbus multimeter to measure and save grid supply quality, every 200ms, since I need to catch millisecond grid drops, and it works, but later Daq the files become too big.

kind regars

MatthewReed303, 2024-10-30:
@mariorl I agree and a progress bar of loading data would help. I have noticed it will also crash the browser session sometimes also. Another possibility is use the new triggers and set a range and log only the ranges you need ignoring the rest. You can also do this on the PLC with a simple FIFO buffer so you don't miss any important and push to Fuxa every 10sec on change etc

mariorl, 2024-11-04:
The instrument I sampling is not a PLC, so can't do FIFO or so to reduce the traffic. What you mean with "new triggers"?

unocelli, 2024-11-06:
Hi, I just added the feature `chunk DAQ history result`, can you give me feedback?

unocelli, 2024-11-13:
[image]

templates on the scripts are only for client functions, this is the reason why for server script is empty

unocelli, 2024-12-11:
I’m going to close this as resolved. let me know if you have any issues.