frangoteam/FUXA #1468: Formula Function

KWYHDHJ, 2024-10-31:
**Describe the feature**
Could you provide a recipe component that allows for batch configuration of multiple tags, supporting batch read and write operations to implement reading and writing operations for multiple tags?

**Describe the solution you'd like**
You can create a list component where each row can bind a label, provide a display column for the current value, and include an input box column for easy subsequent writing operations. Additionally, there should be a confirm button to execute batch writing operations.

**Additional context**
Here is the result after I completed the configuration of a recipe component I designed.
[image]

KWYHDHJ, 2024-11-01:
Can you simply add a list class in the list component that supports writing operations on bound tags

KWYHDHJ, 2024-11-01:
Similar to the effect shown in the screenshot below, the first column has a custom name, the second column has a bound tag, and the third column supports tag writing operations。
[image]

henjoe, 2024-11-01:
+1 for this Feature

It will be better to implement this in the Scripts.

For getTags() and setTags() make the initial parameter to be an array of TagIds.
```
let data = $getTag(['t_dasdssda-24124sd','t_ioweqwe-te1923j'])
// then you can read each individual Tag by
let tag1 = data[0]
let tag2 = data[1]

```

additionally it will be better for getTag to return not just the "value" but the whole Tag object. so we can.

```
let data = $getTag(['t_dasdssda-24124sd','t_ioweqwe-te1923j'])
// then you can read each individual Tag by
let tag1_value = data[0].value;
let tag2 = data[1].value;

tag1_name = data[0].name;
tag2_name = data[1].name;
// And all properties of the Tags can be obtained, even its Daq configuration and etc...
```