frangoteam/FUXA #1475: [FEATURE] SMTP Mailing for Alarms

Kiruthick-Roxx, 2024-11-03:
**Describe the feature**
It would be good feature to add SMTP mailing for individual alarms in the action configuration panel, so that any critical alarms will be immediately notified by mail.

**Describe the solution you'd like**
[image]
In the alarms action property panel, one more option like SMTP can be added and if possible additional settings to send the mail for the alarms can also be added. Upon selecting that, a mail should be sent to the receiver based on the configuration.

**Additional context**
It would be really helpful in the industry as the responsible person will be immediately notified rather than someone noticing the alarm and then taking action. It would also be good if I'm able to store historical alarms to Influxdb..

henjoe, 2024-11-04:
That's what notifications do. 
[image]

Kiruthick-Roxx, 2024-11-05:
But that doesn't work for individual alarms right? Correct me if I'm wrong.

niektuytel, 2024-11-05:
This can been done easily from outside this project.

Kiruthick-Roxx, 2024-11-06:
Can you please share some idea?

unocelli, 2024-11-06:
I suppose that a good compromise could be to add $sendMail function to de script

unocelli, 2024-11-17:
I'm working on this issue $sendMessage

closed 2024-11-21 by Kiruthick-Roxx