frangoteam/FUXA #1479: [BUG]The $getTag() method cannot get the value of the internal data

yy783, 2024-11-05:
**Describe the bug**
Execute the function of opening the card and bind an internal variable A to S7 variable B, click the button to execute the script on the Open Card interface, the input parameter is the internal variable A, the script content is console.log($getTag(a)), and the printed result is undefined.

**To Reproduce**
Steps to reproduce the behavior:
1. Draw a square, set the click event to Open Card, bind an internal variable to an S7 variable
3. There is a button in the interface of the card being opened, and the click event of the button is: execute the test script console.log($getTag(a)), and a is the internal variable bound when the card is opene. The script mode is client
4. Edge browser output is undefined

yy783, 2024-11-11:
The mode I'm using is the client mode, I printed the object with the target id in $getTag, but there is no “value” attribute, so the $getTag() method always returns undefined

unocelli, 2024-11-13:
Hi, I try to understand your description and I am confused to understand what your purpose is

yy783, 2024-11-14:
For example, there is an S7 variable called “LQB1_RunByte”, and there is an internal variable called “runByte”. In the click event of a certain element, it is set to “Open Card” to open “view1” and bind “runByte” and “LQB1_RunByte”. There is a button in “view1”, and when the button is clicked, the "Client-Mode" script "console.log($getTag(runByte))" is executed. The script prints a "null" or "undefined" result. My goal is to create a common card interface for objects of the same type.

unocelli, 2024-12-06:
why do you not use placeholders?

unocelli, 2025-03-04:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2025-03-04 by unocelli