frangoteam/FUXA #1482: [FEATURE] Implement onOpen and onClose Methods for Enhanced View Management

mauroalexandre, 2024-11-05:
**Feature Description**

The goal of this feature is to introduce `onOpen` and `onClose` methods within the View class to manage behaviors linked to the lifecycle of views more effectively. These methods will allow developers to define custom scripts and logic that should run when a view is either opened or displayed to the user, or closed. This enhancement seeks to improve the application's interactivity and responsiveness by allowing more granular control over the view's setup and teardown processes.

**Expected Benefits**

- **Enhanced Flexibility:** Developers can easily manage what happens right before a view is displayed and immediately after it's closed.
- **Improved User Experience:** By managing view states dynamically, the user experience can be made smoother and more intuitive.

**Use Cases**

When user open view we can execute script to check client device type (smartphone, desktop, etc) and redirect to view specific for device layout.

[image]

**Pull Request**
[Pull request here](https://github.com/frangoteam/FUXA/pull/1481).

unocelli, 2024-11-10:
Thanks @mauroalexandre

unocelli, 2024-12-11:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-12-11 by unocelli