frangoteam/FUXA #1489: [BUG] TDengine historian results in unordered points, also data structure in TDengine is poor

alext-extracellular, 2024-11-08:
**Describe the bug**
When TDengine is configured as your DAQ server, reloading the Fuxa page to make graphs show historical data, results in the graphs showing seemingly unordered data from many different tags.
[image]
In this image you can see these messy lines. The Agitation graph should show at 50 constantly, and the pH graph should show at 7.1 roughly. However in this image they have picked up some of each others values and therefore the graph is incorrect. 

Also, the data structure in TDengine created by FUXA is not very useful:
[image]
It puts everything in one supertable, and also the type is not specific. 
The tag name is not kept in the database, so this makes it hard to query the database yourself without using Fuxa:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Add Tdengine as DAQ
2. configure Fuxa to save certain tags from your device ( I am using OPCUA)
3. configure line charts to display these tags
4. let the data record for a while
5. Reload the page
6. view the graph with the wrong data being displayed

**Expected behavior**
Historical data is correctly recalled into the graph without confusion.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment**
 - FUXA version: Version 1.1.19-1524 installed from dockerhub
 - Node.js version: n/a
 - npm version: n/a
 - Platform/OS: linux/docker
 - Browser: thorium

henjoe, 2024-11-09:
Try the updated version 1.2.1 (this could be resolve related to tagid issue- I think someone has already PR this). 

For the data structure, I agree with you, they should have use the STable (like a template table) for each tags rather than joining all tag values in single STable. 
In total fairness to the developer team, they have done a good progress integrating TDEngine, it just that when it comes to optimizing data structure in storing data on a different databases engine requires some sort of expertise for that specific database.

unocelli, 2024-12-11:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2024-12-11 by unocelli