frangoteam/FUXA #1492: [BUG]After running for a few hours, the interface is not refreshing

WuBingo12, 2024-11-10:
After running for a few hours, the real-time table or chart is not refreshing and the pipeline is not running dynamically. Please help analyze the reasons. The client is running in webview2. Clicking the refresh button in the table does not work, but clicking the pop-up dialog box can restore the refresh of the interface, or reloading the page can also restore it 
The image below is not refresh 
[image]
The image below is stop or run very slow
[image]
Click the pop-up dialog box can restore the refresh of the interface
[image]

unocelli, 2024-11-13:
Hi, I suppose that you will show to many records in a html table, for this type of control there are limits

WuBingo12, 2024-11-14:
It feels like after a period of time, no user clicks the trigger, and the angular client enters a dormant state. Now I use a script to regularly trigger the table refresh button on the interface to keep it active, and it has always been normal. But I think this is not a correct solution. Do you have any good suggestions? Is it a memory overflow? Or how to maintain activity normally when there is no user operation?

> Hi, I suppose that you will show to many records in a html table, for this type of control there are limits

unocelli, 2024-11-28:
Hi, which browser are you using? try investigating with browser devtool.

closed 2024-12-11 by unocelli