frangoteam/FUXA #1493: Using bitmask to control "Table Alarms"

robsori, 2024-11-10:
Hi. I tested the new "Table Alarms" control and noticed that it only works if I use binary values ​​for alarms. It would be very useful if we could use "bitmask" to use 8, 16 and 32 bit tags and extract only the binary values ​​of interest for displaying alarms.
Packing binary values ​​for alarms in a 32-bit tag, in the communication between PLC and SCADA, greatly simplifies data communication and saves us from creating 32 tags in SCADA.

[image]

henjoe, 2024-11-12:
Does this options do the same? 
[image]

robsori, 2024-11-13:
> Does this options do the same? [...]

I don't think it's the same thing, in "Table Alarms" I assume it's a filter that calls the tags that trigger the alarms.

robsori, 2024-11-13:
I don't think it's the same thing, in "Table Alarms" I assume it's a filter that calls the tags that trigger the alarms.

unocelli, 2024-11-13:
@robsori I think you are confusing how to show alarms and how they are configured.