frangoteam/FUXA #1501: [FEATURE]

andy8601, 2024-11-16:
How do I hide the buttons in the Header Navigation Bar from specific users? My attempt to modify the permissions in the properties does not work, the navigation side menu is valid, or can I access the specified page directly through the unique URL address, which hides all navigation bars and only one configuration page?
[image]
[image]
[image]

unocelli, 2024-11-20:
I'm working on this feature

unocelli, 2024-12-11:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-12-11 by unocelli