frangoteam/FUXA #1505: [BUG]Enabling 'Only for Editor' on the settings page has not taken effect

yzkqq345, 2024-11-20:
[image]
I found that the '/api/settings' interface did not set 'secureOnlyEditor' to 'runtime.settings.secureOnlyEditor';

I don't understand Node.js, please forgive me if I'm wrong

mauroalexandre, 2024-11-21:
Your token are expiring with this option enabled?

yzkqq345, 2024-11-25:
> Your token are expiring with this option enabled?

No, token is valid
'post /api/settings' did not handle the parameter 'secureOnlyEditor'
[image]
[image]
[image]

yzkqq345, 2024-11-25:
> Your token are expiring with this option enabled?

The parameter 'secureOnlyEditor' is only written to the 'mysettings. json' file, so it needs to be restarted in the Docker container to take effect

unocelli, 2024-11-25:
Hi, some settings require a restart of the backend and this is one of them

yzkqq345, 2024-11-26:
> Hi, some settings require a restart of the backend and this is one of them

So why not add a prompt on the settings page?

robsori, 2024-12-07:
> > Hi, some settings require a restart of the backend and this is one of them
> 
> So why not add a prompt on the settings page?

Hello, did you get positive results with restarting the backend? In my case the token expires in any situation: with or without the "Only for Editor" option. It expires after every 24 hours. I mention that the operator works very rarely on the application. After the token expires, no script works anymore.
[image]

unocelli, 2024-12-11:
Hi, Please let me know more details!

robsori, 2024-12-12:
The philosophy of working with tokens was quite confusing until I understood how it worked. Initially I thought that when the token expired, the backend would also stop working, which would not have been right. We can close this topic. Thank you very much for the support.

closed 2024-12-12 by unocelli