frangoteam/FUXA #1506: [FEATURE] Interactive charting library

Kiruthick-Roxx, 2024-11-21:
**Describe the feature**
It would be great to use Highchart library to display a line chart that will have different color zones like the one below.

[image]

After a lot of try to change the colors line for different zones in Uplot library i found it to be impossible as the library is not designed in such a way or I may not be able to do it. Then i tried out this library in a normal angular project and i was able to do it but unfortunately i couldn't do that in FUXA as I'm very much new to Angular and JS etc. So i request this feature which would be much more than a normal chart as it would give much better representation of data and immediate analysis can be made based on the color zones.

leeoniya, 2024-11-23:
uplot does support this, but you have to use some functions to generate the zone gradients. it does not have config options to do this out of the box. FUXA can definitely offer this to users, though :)

https://leeoniya.github.io/uPlot/demos/gradients.html

henjoe, 2024-11-25:
> uplot does support this, but you have to use some functions to generate the zone gradients. it does not have config options to do this out of the box. FUXA can definitely offer this to users, though :)
> 
> https://leeoniya.github.io/uPlot/demos/gradients.html

I was surprised you are here 😄

unocelli, 2024-11-29:
@Kiruthick-Roxx and @unocelli are working on this feature

unocelli, 2025-03-04:
I’m going to close this as resolved. let me know if you have any issues.

closed 2025-03-04 by unocelli