frangoteam/FUXA #1510: Can sound be added to the alarm?

s5314203, 2024-11-26:
Can sound be added to the alarm? Can the alarm list page be called separately? For example, placing it in an iframe

mauroalexandre, 2024-11-26:
Hello @s5314203 !

To display alarms in other locations, simply add the "Table" component to your View and set the "Type" property to "Alarms table."

Regarding sound customization, I believe it is currently not possible.

@unocelli , what do you think about this feature?

unocelli, 2024-11-26:
Hi, there is a discussion about this #1142 you can solve it via script

Wall-E1271, 2024-12-18:
But I found that after you confirm the alarm, when the process value returns to the normal range again (there is no alarm value at this time), the value of "toack" will change to 1, and then immediately 0 again, which will generate an alarm. I don't know if the FUXA team did this intentionally, I hope it doesn't, thank you!

unocelli, 2025-03-04:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2025-03-04 by unocelli