frangoteam/FUXA #1512: How to retrieve all alarms in the script

s5314203, 2024-11-27:
How to retrieve all alarms in the script

unocelli, 2024-11-29:
I'm working on this feature

unocelli, 2024-12-01:
I just added the feature in master branch

const activeAlarms =  await $getAlarms();
/* result in activeAlarms = [
    {
        "name": "temperature^~^highhigh",
        "type": "highhigh",
        "ontime": 1733074995709,
        "offtime": 1733075016983,
        "acktime": 0,
        "status": "NF",
        "text": "Alarm temperature",
        "bkcolor": "#FF4848",
        "color": "#FFF",
        "toack": 1
    }, {...}
] */

const ackResult = await $ackAlarm('temperature', ['highhigh', 'high']); // to ack 'highhigh' and 'high of alarm temperature
/* result ackResult = [true, ...] */

unocelli, 2024-12-11:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-12-11 by unocelli