frangoteam/FUXA #1516: ODBC and  MySQL BUG!!

dkq08, 2024-11-28:
"I am communicating with MySQL through FUXA's ODBC, and I am receiving error messages! I am deployed on Docker, and this bug is causing me a lot of frustration!"  If you need further assistance, feel free to ask!

Looking forward to your reply.

[image]

MatthewReed303, 2024-11-28:
@dkq08 please show complete connection string and also have a read through this issue, lot's of info in there
 
https://github.com/frangoteam/FUXA/issues/1204
https://github.com/frangoteam/FUXA/wiki/HowTo-ODBC

How is the security setup in your MySQL DB? This is a common issue with MySQL and always seems to be the issue. Just to confirm you can ping the MySQL IP from withing the docker container?

dkq08, 2024-11-29:
I found the problem, which was related to security, but now there is a new error with the following content: "getendpoints-error: Error: [odbc] Error getting table information." 
So far the connection is successful!
Full character:  DRIVER=MySQL;SERVER=192.168.0.33;PORT=3306;DATABASE=testdb

[image]

[image]

MatthewReed303, 2024-11-29:
@dkq08 just ignore that error and try to use sql in the scripting

dkq08, 2024-12-02:
If it is based on Windows, does ODBC need to be installed manually?

closed 2025-03-04 by unocelli