frangoteam/FUXA #1523: mqtt device timestamp[BUG]

haslish, 2024-12-05:
I have a mqtt client and when the registration is disabled the timestamp is correct and shows the time that the mqtt message arrives. 

When registration is enabled the timestamp changes. I looks like the timestamps for the tags with registraton enable gets a new timestamp when a other tag gets a new value until all tags have a value then all timestamps updates with the same rate as the sample time.
I tried and disable registration for one tag and the timestamp for that tag works while the other ones with registration enabled doesnt.

Sample Interval 1s 
Save value interval 7200s

unocelli, 2024-12-06:
Hi, I am not sure I understand, can you provide an example?

haslish, 2024-12-07:
Also the restore function doesnt work with registration enabled.

unocelli, 2025-02-14:
We can not reproduce it, please provide a example project

unocelli, 2025-03-04:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2025-03-04 by unocelli